mccdaq/uldaq #4: Device not found

lmlr, 2018-08-02:
Hi,

I'm trying to use the python API with a USB-TEMP device. It works with the windows version (device shows up and I can read temperatures via Python) but the list returned by get_daq_device_inventory is empty when trying the same thing using Ubuntu 16.04 LTS.

The devices shows up using dmesg.

Any hints on what I'm doing wrong would be appreciated.

jeffreyg3, 2018-08-02:
Hello lmlr,

The UL for Linux does not provide support for the USB-TEMP at this time. For a complete list of supported devices:  https://www.mccdaq.com/PDFs/Manuals/Linux-hw.pdf

Though it may be included in a future release, I do not have a schedule at this time.

However, as you have discovered, it does work using the 3rd party driver.
http://kb.mccdaq.com/KnowledgebaseArticle50608.aspx

Best regards,
Jeff

closed 2018-08-02 by sbazaz

sbazaz, 2020-01-08:
Support for this device was added in version 1.1.0. Please make sure uldaq 1.1.0 or later is installed on your Linux system.

ztrayl3, 2020-01-24:
I am facing the same issue, as I am fully able to communicate between my USB-1608FS and my Raspberry Pi ZeroW through the examples in [this guide](https://www.mccdaq.com/TechTips/TechTip-9.aspx) but the python commands from [this guide](https://www.mccdaq.com/TechTips/TechTip-10.aspx) are giving me a "No DAQ devices found" error.

jeffreyg3, 2020-01-24:
uldaq does not support the USB-1608FS but rather the USB-1608FS-Plus.
The library referred to in the above linked article (https://www.mccdaq.com/TechTips/TechTip-9.aspx) uses https://github.com/wjasper/Linux_Drivers.  
That library does include support for the USB-1608FS

Here is a link to the MCC devices supported with uldaq:
(https://www.mccdaq.com/PDFs/Manuals/Linux-hw.pdf)