mccdaq/uldaq #5: Invalid Re-trigger Count

irowebbn, 2018-08-14:
I am attempting to synchronize a MCC 1608FS-Plus with another device using the TRIG_IN  terminal. I had a version where I started a continuous sample with a pulse ([see here](https://github.com/ukyuav/BC5-datalogger/blob/master/main.cpp#L198)), but now I would like one scan per pulse, with the trigger rearming after each sample.

I am using the SO_RETRIGGER option, but I am running into the following problem. If I put 0 for the `retriggerSampleCount`, I get no data. If I put any positive number, I get an "Invalid re-trigger count" error.  You can see my work [here](https://github.com/ukyuav/BC5-datalogger/blob/paced/main.cpp#L200).

I'm not sure if this affects anything, but I notice that there is no mention of the SO_RETRIGGER option in [/src/usb/ai/AiUsb1608fsPlus.cpp](https://github.com/mccdaq/uldaq/blob/master/src/usb/ai/AiUsb1608fsPlus.cpp) like there are in the files for other devices.

MCC-AE, 2018-08-15:
The USB-1608FS-PLUS hardware design does not support the RETRIGGER mode.   However, if your need is to sample all AI channels exactly once each, per external pulse, then the good news is that the USB-1608FS-PLUS hardware's external clock mode (SO_EXTCLOCK) works in that way.

irowebbn, 2018-08-15:
Seems like a reasonable workaround, thanks!

closed 2018-08-15 by irowebbn