mccdaq/uldaq #11: problem detecting USB-1208FS

torbenator, 2019-05-09:
Hi,

I just began using your code and uldaq is not detecting the USB-1208FS daq (which I see is supported) that is plugged into my computer. I can't seem to find the source of the problem.

My computer is a 2014 Macbook Pro running Mojave. (I was running high Sierra but I updated because I thought that might be the issue)

I believe that I installed uldaq correctly as I followed all of the steps on your installation guide and did not see any errors. I reinstalled to confirm.
I can confirm that my computer recognizes the device. 
I also used this daq last week so I don't think that's the issue either. 

When I run AIn, I get an error saying: "No DAQ device detected". 

[image]

Sorry for the trouble, your code is very well documented and readable but I don't know what to do to get it working on my machine. I'd appreciate any ideas you might have for how to fix this.

Thanks!

jeffreyg3, 2019-05-09:
I believe i see the problem.  It is not the USB-1208FS supported under uldaq. It is the USB-1208FS-PLUS that is supported.
The USB-1208FS is supported by the 3rd party linux drivers.

torbenator, 2019-05-09:
Oh OK. My mistake. Thanks!

closed 2019-05-09 by torbenator

mazerj, 2019-06-14:
Hi Jeff,

Any chance you're the contact person for MCC for the uldaq library?  I'm
wondering if there's any way to get information on response latency for
the USB-1208FS-Plus using the uldaq library? Basically, I'm wondering how
much lag there is when calling something like ulAIn() to get a sample from
the device and if that lag is deterministic?

I'm need to be able to determine the true time at which the a sample is
actually acquired (relative to the host computer's clock) to without about
1ms. The application can tolerate some lag, as long as it's deterministic
and known. But it's critical to be able to know what the lag actually is
and/or what it depends on. And I can't figure out anyway to measure it
automatically (using a laptop w/o a parallel port -- so anything that
generates a trigger to test the timing, would also be a usb device with
it's own lags).

Thanks a lot,
jamie

jeffreyg3, 2019-06-14:
Hi Jamie,
It is preferred if you have questions, that you contact us at info@mccdaq.com<mailto:info@mccdaq.com> or support@mccdaq.com<mailto:support@mccdaq.com>.
So your request can be logged and issued to the appropriate person.

I have performed some benchmarking on the USB-1208FS-Plus, here are my results.

PC Used for UL for Linux:   Dell Core 2 duo 3.0MHz x2, running Debian GNU/Linux9 (stretch), 32 bit
Test app created in GNU C, Codeblocks 16.1, and GTK3+
All input functions were tested 3 times, average shown.
All Output tests were monitored on Oscilloscope and because the timing changes constantly, values are approximated (~).
All input values are averaged of 3 test values.
Your results may vary.

Linux

H/W Model
UL Rev
AIn
AOut
DIn
DBitIn
DOut
DBitOut
CIn
USB-1208FS-PLUS
1.00
1000
~1000
1000
950
~1000
~500
930
The results are in readings/updates per second.

Keep in mind performing these static functions are PC and Linux operating system dependent.
These tests were performed by taking 10 seconds worth of data as fast as possible in a tight loop with nothing else occurring in the loop at the time.
And then dividing the data by 10 to get 1 seconds worth of data.
As it states above, your results may vary.

As stated above, if you have additional questions or comments, please send an email to info@mccdaq.com<mailto:info@mccdaq.com> or support@mccdaq.com<mailto:support@mccdaq.com>.
Best regards,
Jeff Greenberg
Application Engineer

[cid:image001.gif@01CA2236.DC1284E0]
Measurement Computing
10 Commerce Way
Norton, MA 02766
P: (508) 946-5100 Opt 2
F: (508) 946-9500
http://www.mccdaq.com/

http://kernel.montana.edu<https://urldefense.proofpoint.com/v2/url?u=http-3A__kernel.montana.edu&d=DwQFaQ&c=I_0YwoKy7z5LMTVdyO6YCiE2uzI1jjZZuIPelcSjixA&r=fvIcWN5VlRxRonVgPz37IA&m=VNPiG9IBBrJ6qeVGqRZkN3g9lV4iulO0sgykbnK96Mo&s=zjdyPM0yDjNtuIuN1fnOu9g_A1e3THRApL7UQPTS2D4&e=>