mccdaq/uldaq #13: Add the python uldaq library on GitHub

echoix, 2019-06-30:
The uldaq python package on PyPi (https://pypi.org/project/uldaq/) links to this repository and the measurement computing website. However, the python library’s code isn’t found anywhere, except inside the archives that can be downloaded from PyPi. Could there be a way to have that code in this repository, or a separate new one, so it could be possible to submit fixes or issues regarding the python specific code?

Ultimately, it should be great that the C part of the library could be bundled/compiled at the python package's setup, but any help here starts by having the package's code visible ;)

Thanks!

sbazaz, 2019-07-08:
Thank you for your suggestions. We'll consider them for the future releases of the library.

closed 2019-07-08 by sbazaz

toddkarin, 2019-11-01:
And please also add more python examples, e.g. analog in scan, etc.