mccdaq/uldaq #23: Set IEPE settings in EEPROM

delijati, 2020-10-01:
The documentation says:

> In Windows, you can use the Open Layers Control panel to configure the IEPE settings that are used at power up.
> Note:The values that are set using ulAISetConfig and ulAISetConfigDbl are not retained in EEPROM; therefore, they are not used a power up. It is recommended that your program initialize these values when your program runs.

Apparently it is possible to set the settings via windows it would be nice if we could also do it via the ulday Linux library

sbazaz, 2020-11-17:
Since the Windows library uses a custom kernel driver, it is able to detect a USB device when  it is attached to the host and read the IEPE setting from EEPROM and apply it right after the device is enumerated. The Linux  does not use a custom kernel driver, the value specified by the ulAISetConfig function is will be stored in EEPROM and when the ulConnectDaqDevice  is invoked the driver reads the IEPE settings from EEPROM and applies the setting then. If a user needs to apply the IEPE setting at power up then he would need create a Linux daemon to monitor when a device attached to the host by calling ulGetDaqDeviceInventory() and call  ulConnectDaqDevice() when a device is detected

closed 2020-11-17 by sbazaz

delijati, 2020-11-17:
Thanks for that explanation

delijati, 2022-11-16:
Hi me again back at this ticket :) When i tried the usb daemon as suggested [1] . I get unfortunately a [2] "Interface claimed" when i try to connect to with my real program . Is there some other thing i need to do!?
[1] Daemon
```python
from pyudev import Context, Monitor
from uldaq import (
    get_daq_device_inventory,
    DaqDevice,
    InterfaceType
)

ctx = Context()
monitor = Monitor.from_netlink(ctx)
monitor.filter_by(subsystem='usb')

interface_type = InterfaceType.ANY
con = False
for device in iter(monitor.poll, None):
    print(
        f"{device.action=}, {device.get('ID_VENDOR_ID')=}, {device.get('ID_MODEL_ID')=}")
    if device.action == "add":
        if not con and device.get("ID_VENDOR_ID") == "0867" and device.get("ID_MODEL_ID") == "9839":
            devices = get_daq_device_inventory(interface_type)
            daq_device = DaqDevice(devices[0])
            daq_device.connect(connection_code=0)
            con = True
            print("connected")
    if device.action == "remove":
        if con and device.get("ID_VENDOR_ID") == "0867" and device.get("ID_MODEL_ID") == "9839":
            con = False
            print("removed")
```
[2] Other program
```zsh
❯ .env/bin/python a_in_scan_iepe.py
Found 1 DAQ device(s):
  [0] DT9837ABC (INTERFACE CLAIMED)

Please select a DAQ device, enter a number between 0 and 0: 0
Traceback (most recent call last):
  File "/opt/develop/mcc-daq/a_in_scan_iepe.py", line 248, in <module>
    main()
  File "/opt/develop/mcc-daq/a_in_scan_iepe.py", line 76, in main
    daq_device = DaqDevice(devices[descriptor_index])
  File "/opt/develop/mcc-daq/.env/lib/python3.10/site-packages/uldaq/daq_device.py", line 42, in __init__
    raise ULException(ULError.BAD_DESCRIPTOR)
uldaq.ul_exception.ULException: ULError.BAD_DESCRIPTOR: Invalid descriptor
Exception ignored in: <function DaqDevice.__del__ at 0x7f976069c3a0>
Traceback (most recent call last):
  File "/opt/develop/mcc-daq/.env/lib/python3.10/site-packages/uldaq/daq_device.py", line 86, in __del__
    self.release()
  File "/opt/develop/mcc-daq/.env/lib/python3.10/site-packages/uldaq/daq_device.py", line 437, in release
    raise ULException(err)
uldaq.ul_exception.ULException: ULError.BAD_DEV_HANDLE: Invalid device handle
```