mccdaq/uldaq #25: C compiler does not work on macOS

garrettj403, 2021-02-23:
I'm trying to install ``uldaq`` on my MacBook Pro (running macOS Catalina, version 10.15.7).

It says that my C compiler does not work when I try to run ``./configure && make`` inside the ``libuldaq-1.2.0`` directory:

```
checking for a BSD-compatible install... /usr/local/bin/ginstall -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/local/bin/gmkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking for style of include used by make... GNU
checking for gcc... x86_64-apple-darwin13.4.0-clang
checking whether the C compiler works... no
configure: error: in `/Users/jgarrett/Projects/libuldaq-1.2.0':
configure: error: C compiler cannot create executables
See `config.log' for more details
```
I do also have ``gcc`` installed with Homebrew. Is it possible to tell ``configure`` to use this gcc installation instead of Apple's clang?

garrettj403, 2021-02-23:
Yes, I followed the instructions to install the prerequisites.

sbazaz, 2021-02-23:
What message do you see when you run gcc from command line?

*$*gcc

garrettj403, 2021-02-23:
If I run ``gcc --version``, I get:
```
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/4.2.1
Apple clang version 12.0.0 (clang-1200.0.32.29)
Target: x86_64-apple-darwin19.6.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

sbazaz, 2021-02-23:
What do you get if you run gcc without any parameters?

garrettj403, 2021-02-23:
> What do you get if you run gcc without any parameters?

```
clang: error: no input files
```

sbazaz, 2021-02-23:
Can you send me your config.log file?

garrettj403, 2021-02-23:
``config.log``:
```
This file contains any messages produced by compilers while
running configure, to aid debugging if configure makes a mistake.

It was created by libuldaq configure 1.2.0, which was
generated by GNU Autoconf 2.69.  Invocation command line was

  $ ./configure 

## --------- ##
## Platform. ##
## --------- ##

hostname = XXXXX
uname -m = x86_64
uname -r = 19.6.0
uname -s = Darwin
uname -v = Darwin Kernel Version 19.6.0: Tue Nov 10 00:10:30 PST 2020; root:xnu-6153.141.10~1/RELEASE_X86_64

/usr/bin/uname -p = i386
/bin/uname -X     = unknown

/bin/arch              = unknown
/usr/bin/arch -k       = unknown
/usr/convex/getsysinfo = unknown
/usr/bin/hostinfo      = Mach kernel version:
	 Darwin Kernel Version 19.6.0: Tue Nov 10 00:10:30 PST 2020; root:xnu-6153.141.10~1/RELEASE_X86_64
Kernel configured for up to 8 processors.
4 processors are physically available.
8 processors are logically available.
Processor type: x86_64h (Intel x86-64h Haswell)
Processors active: 0 1 2 3 4 5 6 7
Primary memory available: 16.00 gigabytes
Default processor set: 350 tasks, 1670 threads, 8 processors
Load average: 3.95, Mach factor: 4.23
/bin/machine           = unknown
/usr/bin/oslevel       = unknown
/bin/universe          = unknown

PATH: /Users/jgarrett/.dotfiles/git/bin
PATH: /Users/jgarrett/opt/anaconda3/bin
PATH: /Users/jgarrett/opt/anaconda3/condabin
PATH: /usr/local/bin
PATH: /usr/bin
PATH: /bin
PATH: /usr/sbin
PATH: /sbin
PATH: /Library/TeX/texbin
PATH: /opt/X11/bin

## ----------- ##
## Core tests. ##
## ----------- ##

configure:2414: checking for a BSD-compatible install
configure:2482: result: /usr/local/bin/ginstall -c
configure:2493: checking whether build environment is sane
configure:2548: result: yes
configure:2699: checking for a thread-safe mkdir -p
configure:2738: result: /usr/local/bin/gmkdir -p
configure:2745: checking for gawk
configure:2761: found /usr/local/bin/gawk
configure:2772: result: gawk
configure:2783: checking whether make sets $(MAKE)
configure:2805: result: yes
configure:2834: checking whether make supports nested variables
configure:2851: result: yes
configure:2988: checking for style of include used by make
configure:3016: result: GNU
[... 220 lines cut ...]
pdfdir='${docdir}'
prefix='NONE'
program_transform_name='s,x,x,'
psdir='${docdir}'
sbindir='${exec_prefix}/sbin'
sharedstatedir='${prefix}/com'
sysconfdir='${prefix}/etc'
target_alias=''

## ----------- ##
## confdefs.h. ##
## ----------- ##

/* confdefs.h */
#define PACKAGE_NAME "libuldaq"
#define PACKAGE_TARNAME "libuldaq"
#define PACKAGE_VERSION "1.2.0"
#define PACKAGE_STRING "libuldaq 1.2.0"
#define PACKAGE_BUGREPORT "info@mccdaq.com"
#define PACKAGE_URL "http://www.mccdaq.com"
#define PACKAGE "libuldaq"
#define VERSION "1.2.0"

configure: exit 77
```

sbazaz, 2021-02-23:
The issue might be related to *xcode command line tools 12* installed on
your system. Please uninstall it by running the following command

$ sudo rm -rf /Library/Developer/CommandLineTools

Download *Command Line Tools for Xcode 11.5* from here
<https://developer.apple.com/download/more/> (Dated: May 20, 2020) and
install it on your system

garrettj403, 2021-02-25:
I completely reinstalled X-code and it's working now. (Just reinstalling CommandLineTools didn't work.)

Thank you for your help!

closed 2021-02-25 by garrettj403