mccdaq/uldaq #29: Channels in `a_in_load_queue()` must be in ascending order?

MarinManuel, 2021-07-08:
Is there a way to pass an arbitrary list of `AiQueueElement` to `a_in_load_queue()` where the channels are not necessarily in order?

Consider the following code. Tested with `uldaq==1.2.2` and USB-201

```python
from uldaq import (get_daq_device_inventory, DaqDevice, AInScanFlag,
                   AiInputMode, AiQueueElement, create_float_buffer,
                   ScanOption, InterfaceType, Range)

chan_list = [0, 1]
bufferSz = 5000
rate = 100
scan_options = ScanOption.DEFAULTIO | ScanOption.CONTINUOUS
flags = AInScanFlag.DEFAULT

devices = get_daq_device_inventory(InterfaceType.ANY)
daq_device = DaqDevice(devices[0])
ai_device = daq_device.get_ai_device()
daq_device.connect(connection_code=0)
ai_info = ai_device.get_info()
queue_list = []
for chan in chan_list:
    queue_element = AiQueueElement()
    queue_element.channel = chan
    queue_element.input_mode = AiInputMode.SINGLE_ENDED
    queue_element.range = Range.BIP10VOLTS
    queue_list.append(queue_element)

ai_device.a_in_load_queue(queue_list)
data = create_float_buffer(len(chan_list), bufferSz)
rate = ai_device.a_in_scan(0, 0, AiInputMode.SINGLE_ENDED, Range.UNI10VOLTS,
                           bufferSz,
                           rate, scan_options, flags, data)
status, transfer_status = ai_device.get_scan_status()
```

This code works fine,
but inverting the order of the channels
`chan_list = [1, 0]`
raises an Exception:

> ```text
>Traceback (most recent call last):
>  File "(...)", line 26, in <module>
>   ai_device.a_in_load_queue(queue_list)
>  File "(...)/lib/python3.9/site-packages/uldaq/ai_device.py", line 150, in a_in_load_queue
>    raise ULException(err)
>uldaq.ul_exception.ULException: ULError.BAD_AI_CHAN_QUEUE: Invalid channel queue specified
>```

jeffreyg3, 2021-07-08:
Sorry no, for the USB_200 series the channel-gain queue is limited to 8 elements. Channels must be listed in ascending order; duplicate channels are not allowed.

MarinManuel, 2021-07-08:
Thanks for the info. Is that indicated anywhere in the docs?

jeffreyg3, 2021-07-08:
it is, MCC has two help files.  The one containing caveats like this is https://www.mccdaq.com/pdfs/manuals/Mcculw_WebHelp/ULStart.htm
This is for Windows based drivers and libraries but not Linux.  When MCC moved to supporting Linux, it was decided User's should look at https://www.mccdaq.com/PDFs/Manuals/UL-Linux/c/index.html for C/C++, 
and https://www.mccdaq.com/PDFs/Manuals/UL-Linux/python/index.html for Python.

In an effort to not duplicate device/library considerations (like queue packing) in documentation, it was decided to only provide that documentation in the https://www.mccdaq.com/pdfs/manuals/Mcculw_WebHelp/ULStart.htm

Sorry if there was any confusion.

closed 2021-07-08 by MarinManuel