mccdaq/uldaq #30: Different results on different computers

toddkarin, 2021-08-08:
I have installed uldaq==1.2.2 in a python virtual environment. I ran the following script on two computers:

```python
"""
Repetitive data logger for home monitoring system.

Now using 18-bit DAQ, so just storing data as float 32.

"""

import numpy as np

from uldaq import (get_daq_device_inventory, DaqDevice, InterfaceType,
                   AiInputMode, Range, AInFlag, AInScanFlag,
                   create_float_buffer,
                   ScanOption)

# Get a list of available DAQ devices
devices = get_daq_device_inventory(InterfaceType.USB)
# Create a DaqDevice Object and connect to the device
daq_device = DaqDevice(devices[0])
daq_device.connect()
# Get AiDevice and AiInfo objects for the analog input subsystem
ai_device = daq_device.get_ai_device()
ai_info = ai_device.get_info()

# Sample acquisition rate in Hz
rate = 200000

# Number samples per channel.
samples_per_channel = int(3/60*rate)

# Number of channels, must agree with specification of low_channel and
# high_channel below
num_channels = 1
low_channel = 0
high_channel = 0

print('Waiting to acquire.')

buffer = create_float_buffer(num_channels, samples_per_channel)

actual_scan_rate = ai_device.a_in_scan(
    low_channel=low_channel,
    high_channel=high_channel,
    input_mode=AiInputMode.DIFFERENTIAL,
    analog_range=Range.BIP5VOLTS,
    samples_per_channel=samples_per_channel,
    rate=rate,
    options=ScanOption.DEFAULTIO,
    flags=AInScanFlag.DEFAULT,
    data=buffer)

# Sleep until the scan is over.
print('Acquiring Scan Rate: {}'.format(actual_scan_rate))
ai_device.scan_wait(1,-1)

# Get data
data = np.array(buffer[:]).reshape(
    (samples_per_channel, num_channels)).transpose()

print(data)

daq_device.disconnect()
daq_device.release()

# Plot data
import matplotlib
matplotlib.use('TKagg')
import matplotlib.pyplot as plt

plt.figure(0)
plt.clf()
cax = np.arange(samples_per_channel)
t = 1 / actual_scan_rate * np.arange(samples_per_channel)
plt.plot(t, data[0,:])
plt.show()
plt.savefig('macbookpro_scan.png')

```

When I use a Macbook Pro, I get this result (this is the line to line voltage in my house):

[image]

However, when I run the same script on a Raspberry Pi, using exactly the same DAQ setup, I get a very different result: 
[image]

Any ideas why this would be, or how to fix?

jeffreyg3, 2021-08-09:
Please try this example:  https://kb.mccdaq.com/KnowledgebaseArticle50760.aspx

The app/script writes the data to file.  The data can then be plotted using LibreOffice Calc or other spreadsheet app.

The reason I'm asking you to test this other app is because this app does not use numpy nor matplotlib.  We have seen behavioral issues when these libs are used on different platforms.  The above linked app does not use them.  

another example app would be https://kb.mccdaq.com/KnowledgebaseArticle50775.aspx
this is an 8 channel logger (high speed), but was written in C and has built in plotting and logging features.  You will need to follow the instructions found in the readme.txt (found at the link).

toddkarin, 2021-08-09:
I ran the first example code (with a few small modifications shown below) using two computers (MacBook Pro and Raspberry Pi). They still give the different results, see the time traces here:

[image]

The raw files are here:
[textfile_rpi.csv](https://github.com/mccdaq/uldaq/files/6956230/textfile_rpi.csv)
[textfile_macbook_pro.csv](https://github.com/mccdaq/uldaq/files/6956231/textfile_macbook_pro.csv)

And the code that was run:

```python
#!/usr/bin/env python
# -*- coding: UTF-8 -*-

"""
    Wrapper call demonstrated:        ai_device.a_in_scan()

    Purpose:                          Performs a continuous scan of the range
                                      of A/D input channels

    Demonstration:                    Displays the analog input data for the
                                      range of user-specified channels using
                                      the first supported range and input mode

    Steps:
    1. Call get_daq_device_inventory() to get the list of available DAQ devices
    2. Create a DaqDevice object
    3. Call daq_device.get_ai_device() to get the ai_device object for the AI subsystem
    4. Verify the ai_device object is valid
    5. Call ai_device.get_info() to get the ai_info object for the AI subsystem
    6. Verify the analog input subsystem has a hardware pacer
    7. Call daq_device.connect() to establish a UL connection to the DAQ device
    8. Call ai_device.a_in_scan() to start the scan of A/D input channels
    9. Call ai_device.get_scan_status() to check the status of the background operation
    10. Display the data for each channel
    11. Call ai_device.scan_stop() to stop the background operation
    12. Call daq_device.disconnect() and daq_device.release() before exiting the process.
"""
from __future__ import print_function
from time import sleep, time
from os import system
from sys import stdout

from uldaq import get_daq_device_inventory, DaqDevice, AInScanFlag, ScanStatus, \
    ScanOption
from uldaq import create_float_buffer, InterfaceType, AiInputMode, Range

def main():
    daq_device = None
    ai_device = None
    status = ScanStatus.IDLE

    descriptor_index = 0
    range_index = 0
    interface_type = InterfaceType.USB
    low_channel = 0
    high_channel = 0
    samples_per_channel = 20480
    buffer_mid_point = int(samples_per_channel / 2)
    buff_check = 0  # 0 = lowerhalf, 1 = upperhalf
    rate = 10240
    scan_options = ScanOption.CONTINUOUS
    flags = AInScanFlag.DEFAULT
    # input_mode = AiInputMode.DIFFERENTIAL,
    # analog_range = Range.BIP5VOLTS,

    try:
        # Get descriptors for all of the available DAQ devices.
        devices = get_daq_device_inventory(interface_type)
        number_of_devices = len(devices)
[... 139 lines cut ...]
                ai_device.scan_stop()
            if daq_device.is_connected():
                daq_device.disconnect()
            daq_device.release()
            f.close()

def display_scan_options(bit_mask):
    options = []
    if bit_mask == ScanOption.DEFAULTIO:
        options.append(ScanOption.DEFAULTIO.name)
    for so in ScanOption:
        if so & bit_mask:
            options.append(so.name)
    return ', '.join(options)

def reset_cursor():
    stdout.write('\033[1;1H')

def clear_eol():
    stdout.write('\x1b[2K')

if __name__ == '__main__':
    main()

```

jeffreyg3, 2021-08-09:
I think you may have a grounding issue (MacBook vs the RPi).  Attached is the example and the data file. Note I used a function generator to create a 60Hz sine wave, 5 V peak
[a_in_scan_to_file.zip](https://github.com/mccdaq/uldaq/files/6956795/a_in_scan_to_file.zip)

[image]

toddkarin, 2021-08-09:
Interesting, how would you suggest fixing this grounding issue? The measurement circuit I am using is 

L1 -- R1 -- R2 -- R3 -- L2 

where L1 is hot phase 1, L2 is hot phase 2, and R1,R2,R3 are resistors (R1=R3). I am measuring the voltage across R2 (this is sent to the DAQ) to get a measure of the L1-L2 voltage.

toddkarin, 2021-08-09:
I changed the measurement circuit. Now I am using a 240 VAC / 24 VAC transformer, and a 2 KOhm, 20 KOhm voltage divider on the secondary. You can see the comparison of the different files here:

So we see that we still have the same issue, even when removing all ground problems with galvanic isolation!

[image]

toddkarin, 2021-08-10:
It turned out to be a grounding issue. Once I grounded the secondary of the transformer to AGND, the problem is fixed!

[image]

closed 2021-08-10 by sbazaz