mccdaq/uldaq #32: OSError: libuldaq.so : cannot open shared object file: No such file or directory

DSMaryam, 2021-10-05:
I followed the extended installation procedure to include python uldaq library, and stumbled on this error while trying to launch a .py file (from provided examples).

File "usr/lib/python2.7/ctypes/__init__.py", line 366, in __init__
self._handle = _dlopen(self._name,mode)
OSError: libuldaq.so : cannot open shared object file: No such file or directory

jeffreyg3, 2021-10-05:
Try launching the script with Python3 like this:

$ python3 ./a_in_scan.py

DSMaryam, 2021-10-05:
Hi, thanks for the quick response. However, it is not working. I have tried ldconfig and many other options found on other issues but none works.

jeffreyg3, 2021-10-05:
What does it respond with when you entered "python3 ./a_in_scan"  ?

What MCC device are you using?

Can you launch one of the compiled C examples?

for example:
./AIn

DSMaryam, 2021-10-05:
"python3 ./a_in_scan"  gives me No such file or directory, "python3 ./a_in_scan.py" gives me the reported error.

I use USB-1408FS-Plus

./AIn works perfectly fine. Problems arise when I try to launch python files.

jeffreyg3, 2021-10-05:
if you get an error stating  "usr/lib/python2.7/ctypes/init.py" then you are running python2.x, and not running python3.x.
Python 2._any_ went obsolete January 2020.

I too have a USB-1408FS-PLUS (among many other MCC devices)

I am running a Raspberry Pi 4B, 
Version = 10 (buster)
with Python 3.7

When I run "python3 ./a_in_scan.py"
it runs and returns data.

I'm not getting reports from any other customers.
Perhaps you have something not installed correctly.

DSMaryam, 2021-10-05:
I too am using Rasberry Pi 4B and moved to using Python3 to launch the files. Still not working...

DSMaryam, 2021-10-06:
Eventually found a way to make it work:
- I opened Root Terminal (or I had permission denied to the device and sudo+command wouldn't work)
- I ran command "  export LD_LIBRARY_PATH=path_to_libuldaq_library/src/.libs  "
- I launched the file without problem

closed 2021-10-06 by DSMaryam

sbazaz, 2021-10-06:
That's odd! Do you have Raspberry PI OS installed on  your RPI 4B?

ericavarga, 2023-03-24:
Hi, I am having similar issues with uldaq installation. I keep getting this error when I try importing it in python "FileNotFoundError: Could not find module 'libuldaq.so' (or one of its dependencies). Try using the full path with constructor syntax." However, it is in my usr/local/lib folder, which seems to be the correct place for it. I followed the download instructions on the github page for mccdaq and uldaq, and I am running Linux on my Windows machine. Any idea why this might be happening?