mccdaq/uldaq #33: undefined reference to `ulGetDaqDeviceInventory'

liyunfei1994, 2021-11-01:
hello~

I installed the uldaq on my computer (**Ubuntu 18.04**)

when I want to compile the .c file, the following errors occured...
```
gcc test.c -o test
```
```
/tmp/ccLwGb0k.o: In function `main':
test.c:(.text+0x66): undefined reference to `ulGetDaqDeviceInventory'
test.c:(.text+0x9f): undefined reference to `ulCreateDaqDevice'
test.c:(.text+0xca): undefined reference to `ulConnectDaqDevice'
test.c:(.text+0x10b): undefined reference to `ulAIn'
test.c:(.text+0x151): undefined reference to `ulDisconnectDaqDevice'
test.c:(.text+0x160): undefined reference to `ulReleaseDaqDevice'
collect2: error: ld returned 1 exit status
```
my question is 
**how to compile and execute an .exe file correctly?**

Thanks!
Looking forward to receiving your replies...

sbazaz, 2021-11-01:
Your gcc command is missing the linker flag. You need to use -l flag to specify the name of the library you program is linked against. In this case the library name is uldaq

**gcc test.c  -luldaq -o test**

closed 2021-11-01 by sbazaz

vdcHyperion, 2022-08-30:
I'm having the same problem, the given solution works the way it says but when we try to run in debug mode these undefined references just pop up, no matter how we set the path for the includes . ... How could it be resolved?