mccdaq/uldaq #39: Problems installing MCC uldaq on a Mac

ajw36, 2022-04-14:
I'm trying to install mccdaq/uldaq on an iMac (Intel) running Monterey (v 12.1).
I've followed the instructions at https://github.com/mccdaq/uldaq.
When I get to './configure && make' I get the following errors:

Making all in src
  CXXLD    libuldaq.la
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: .libs/libuldaq.a(hid_linux.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: .libs/libuldaq.a(UlCtrConfig.o) has no symbols
ranlib: object: .libs/libuldaq.a(CtrInfo.o) malformed object (unknown load command 1)
make[2]: *** [libuldaq.la] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2

Ive been unable to resolve this by internet searches. Would be very grateful if anyone is able to help.
Many thanks. Alex.

jeffreyg3, 2022-04-14:
We had another customer with possibly the same problem.  Here is how he was able to resolve it:

ran the following commands:

which gcc
/opt/local/bin/gcc
gcc --version
gcc (MacPorts gcc6 6.5.0_5) 6.5.0

This indicated that the C compiler was tied to MacPorts, which apparently had not updated since I had updated my operating system. Following the steps listed here, I was able to update MacPorts and resource gcc to XCode by default. Running those same commands after a terminal restart now gave:

which gcc
/usr/bin/gcc
gcc --version
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/4.2.1
Apple clang version 13.0.0 (clang-1300.0.29.30)
Target: x86_64-apple-darwin20.6.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin

This confirmed that the C compiler was being sourced from XCode. I then ran ./configure && make and was able to compile it successfully!
--------------------------------------
Please let me know if helped.
Regards,
Jeff

jeffreyg3, 2022-04-19:
@ajw36 I'm looking for some closure here, Did this resolve your issue?

ajw36, 2022-04-20:
Dear Jeff,
Thanks for your reply and sorry for late response - holiday!
I'm working on this now - not sure if the problem is the same. The first commands give:
which gcc
/usr/bin/gcc
gcc --version
Configured with: --prefix=/Library/Developer/CommandLineTools/usr --with-gxx-include-dir=/Library/Developer/CommandLineTools/SDKs/MacOSX12.1.sdk/usr/include/c++/4.2.1
Apple clang version 13.0.0 (clang-1300.0.27.3)
Target: x86_64-apple-darwin21.2.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin

In your previous post you say "Following steps listed here ...", and I wondered if there is meant to be a link?

ajw36, 2022-04-20:
Dear Jeff,
Just to be clear: the problem I'm still having is knowing how to 'resource gcc to XCode by default'
Many thanks,
Alex

jeffreyg3, 2022-04-20:
Hi Alex,
I've asked around here, we are not sure why you are having that issue.
One suggestion is that you uninstall XCode, then reinstall it from the app store.

I also found this link when I Googled 'how to resource gcc to XCode by default':
https://stackoverflow.com/questions/61876572/set-the-default-gcc-to-xcode-command-line-tools

jeffreyg3, 2022-05-27:
A solution to this issue has been posted here:  https://forum.digilent.com/topic/23095-uldaq-libusb-problem-on-m1-mac/