mccdaq/uldaq #44: libusb problem on M1 Mac

etjensen, 2022-04-26:
I cannot build the library on an M1 Mac- ./configure gives an error:
```
checking operating system... arm-apple-darwin21.4.0
 Mac OS X
checking for libusb_init in -lusb-1.0... no

  Library libusb-1.0 was not found on this system.
  Please install it and re-run ./configure
```

even though I have installed libusb using homebrew and it claims that it is a universal build.

config.log says:
`configure:16180: checking for libusb_init in -lusb-1.0
configure:16205: gcc -o conftest -O3   conftest.c -libusb-1.0   -framework IOKit -framework CoreFoundation >&5
ld: library not found for -libusb-1.0
clang: error: linker command failed with exit code 1 (use -v to see invocation)
configure:16205: $? = 1
configure: failed program was:
| /* confdefs.h */
`

The build on my x86 iMac seemed to work fine.

jeffreyg3, 2022-04-26:
Hello,
unfortunately, uldaq is not currently compatible with MacBook Pro with M1 apple silicon.
At this time, I do not have a timeframe for when it will be.

Sorry for the inconvenience.

jeffreyg3, 2022-04-26:
see also #43 #27

jeffreyg3, 2022-05-02:
This worked for me:

Download the source from github.com:
https://github.com/libusb/libusb/releases/tag/v1.0.26

Build it in the standard way:

$ tar xvfj libusb-1.0.26.tar.bz2
$ cd libusb-1.0.26
$ ./configure
$ make
$ make check
$ sudo make install

Then you can continue on with uldaq installation at ./configure && make

etjensen, 2022-05-03:
Oddly I still get the same error on ./configure- possibly not connecting to the right "libusb"? I am stumped by this.
`checking how to hardcode library paths into programs... immediate
checking operating system... arm-apple-darwin21.4.0
 Mac OS X
checking for libusb_init in -lusb-1.0... no

  Library libusb-1.0 was not found on this system.
  Please install it and re-run ./configure`

jeffreyg3, 2022-05-03:
no promises but something else you could try, would be to install MacPorts
https://www.macports.org/install.php

While I was working on the solution on my MacBook Pro 14, I had installed this along the way.

sbazaz, 2022-05-03:
@etjensen could you please check your /opt/homebrew/lib/ folder to see if the libusb-1.0.0.dylib file is installed there?

birthdaycorp, 2023-03-24:
> @etjensen could you please check your **/opt/homebrew/lib/** folder to see if the **libusb-1.0.0.dylib** file is installed there?

correct me if i'm wrong, but i don't think **/opt/homebrew/lib** exists on M1 or M2 macs. or even have a reason to exist? i have an M2, and when i run brew list libusb, i get the following:

```
/usr/local/lib
/usr/local/opt/libusb/lib
/usr/local/Cellar/libusb/1.0.26/include/libusb-1.0/libusb.h
/usr/local/Cellar/libusb/1.0.26/lib/libusb-1.0.0.dylib
/usr/local/Cellar/libusb/1.0.26/lib/pkgconfig/libusb-1.0.pc
/usr/local/Cellar/libusb/1.0.26/lib/ (2 other files)
/usr/local/Cellar/libusb/1.0.26/share/libusb/ (9 files)
```

granted, my need for **libusb-1.0.0.dylib** has nothing to do with uldaq, but the core concerns are overlapping

jaschawilcox, 2023-09-05:
I was able to get past this by adding some additional arguments pointing directly at homebrew:

`$./configure LDFLAGS="-L/opt/homebrew/lib" CPPFLAGS="-I/opt/homebrew/include"`

Then we can make normally
`$make`

larer, 2024-07-09:
Hi, I'm new here, so bear with me.
I have a Mac M2, with MacPorts, I got the same problem but got past the configure problem by pointing the flags to:
`./configure LDFLAGS="-L/opt/local/lib" CPPFLAGS="-I/opt/local/include"`
Hope this helps someone.

Then I got some 27 warnings during
`make`
Most of them on "anonymous non-C-compatible type..." when void is used,
some other were  "'kIOMasterPortDefault' is deprecated:" and  "'sprintf' is deprecated:"