mccdaq/uldaq #46: can we use in k8s pod?

anup28, 2022-05-29:


anup28, 2022-05-30:
We are using USB-1024LS , can see in lsusb output. But getting error as <Error: No DAQ devices found>

anup28, 2022-05-30:
output of lsusb::: 

<Bus 009 Device 002: ID 09db:0076 MCC USB-1024LS
Bus 009 Device 001: ID 1d6b:0002 Linux 5.4.0-113-generic xhci-hcd xHCI Host Controller
Bus 010 Device 001: ID 1d6b:0003 Linux 5.4.0-113-generic xhci-hcd xHCI Host Controller
Bus 002 Device 001: ID 1d6b:0002 Linux 5.4.0-113-generic ehci_hcd EHCI Host Controller
Bus 008 Device 001: ID 1d6b:0001 Linux 5.4.0-113-generic uhci_hcd UHCI Host Controller
Bus 007 Device 001: ID 1d6b:0001 Linux 5.4.0-113-generic uhci_hcd UHCI Host Controller
Bus 006 Device 001: ID 1d6b:0001 Linux 5.4.0-113-generic uhci_hcd UHCI Host Controller
Bus 001 Device 001: ID 1d6b:0002 Linux 5.4.0-113-generic ehci_hcd EHCI Host Controller
Bus 005 Device 001: ID 1d6b:0001 Linux 5.4.0-113-generic uhci_hcd UHCI Host Controller
Bus 004 Device 001: ID 1d6b:0001 Linux 5.4.0-113-generic uhci_hcd UHCI Host Controller
Bus 003 Device 001: ID 1d6b:0001 Linux 5.4.0-113-generic uhci_hcd UHCI Host Controller
>

samkristoff, 2022-06-01:
Hey @anup28,

I think this should be possible if you configure the container correctly:

https://stackoverflow.com/questions/42697985/kubernetes-node-device-port-usb-mapping-to-pod-or-swarm-service-device-mapp

Out of curiosity, what are you working on needs to access the MCC hardware from K8s?

thompcd, 2023-04-18:
I was able to run this out of a docker container, so I assume you could do it with a cluster, although I'm also curious as to why somebody would do that.. On the host device, you'll need to map the host device inside of the container as referenced above, but also you'll need to modify the `install.sh` script. It calls `daqhats_read_eeprom` during the setup as a validation and will abort on failure, but this will never pass during your makefile execution since it requires host hardware not in place yet.

I broke out the install script to just the compilation & installation and created a startup script that will execute the `daqhats_read_eeprom` on boot to ensure the hats stay up to date.