mccdaq/uldaq #49: Issue with hid_flush_input_pipe()

malapatiravi, 2022-09-19:
Dear Team,
I am using uldaq inside docker container. 
When i stop and start the container i am seeing the following error
[image]
Did any of you face this issue?
After rebooting the host machine multiple times solves this issue. this issue even comes up when the device is physically unplugged too. 
Is there a way we can solve this so that we don't need manual intervention to manually reboot machine multiple times?

Thanks.

jeffreyg3, 2022-09-19:
The uldaq library and related products were never tested inside a docker container, and there are no plans at this time.

sbazaz, 2022-10-14:
The responsibility of the **hid_flush_input_pipe** function is to clear the input pipe buffer if the previous process crashed and there is stale data in the buffer from the previous operation.  You can comment out the line that [invokes](https://github.com/mccdaq/uldaq/blob/24ba881e81ceeb24ff924852c7f56dda072d0282/src/hid/HidDaqDevice.cpp#L76) hid_flush_input_pipe and rebuilt and reinstall the library. However, if your application crashes for any reason or if you use ctrl+c  to terminate your application then you might have to unplug and reconnect your DAQ device before running the application again.

closed 2023-04-20 by sbazaz