mccdaq/uldaq #53: configure && make error

jayebbs98, 2023-03-23:
When running ./configure && make I see the following two errors: 

        In file included from **UlDaqDeviceManager.cpp:11:0:
        ./usb/UsbDaqDevice.h:136:84: error: 'libusb_hotplug_event'** has not been declared
          static int hotplugCallback(struct libusb_context* ctx, struct libusb_device* dev, libusb_hotplug_event event, void *user_data);
                                                                                            ^
        **./usb/UsbDaqDevice.h:149:9: error: 'libusb_hotplug_callback_handle'** does not name a type
          static libusb_hotplug_callback_handle mHotplugHandle;

Has anyone else experienced the same thing or have any suggestions? I tried the latest and oldest versions and get the same error.

FaSoa, 2023-03-23:
Hello @jayebbs98 .
Are just following the Readme to build and install the library?  Which OS are you using?

jayebbs98, 2023-03-23:
Yes I am following the Readme, using CentOS

FaSoa, 2023-03-23:
Any errors when you ran the prerequisite commands?
   $ sudo yum install gcc gcc-c++ make
   $ sudo yum install libusbx-devel

jayebbs98, 2023-03-24:
No errors with the gcc and libusb commands but the libusb command does return the following: 
"Package libusbx-devel-1.0.15-4.el7.x86_64 already installed and latest version
Nothing to do"
And I'm fairly certain 1.0.15 is NOT the latest version...

jayebbs98, 2023-03-24:
This is a libusb issue, not a uldaq issue

closed 2023-03-24 by jayebbs98