mccdaq/uldaq #55: USB-TEMP config issue

thojDufour, 2023-05-23:
Apparently it does not seem to be possible to change the sensor type with the `ulAISetConfig(DaqDeviceHandle daqDeviceHandle, AiConfigItem configItem, unsigned int index, long long configValue)` function.

The only way I found to set it is to use a VM with Windows, and change the config with InstaCal.

To try this, I set all the channels as Thermocouple in InstaCal, then I use this code to set some channels as RTD SCT_3_WIRE:
```C lang
if (ulAISetConfig(daqDeviceHandle[devN], AI_CFG_CHAN_TYPE, i, AI_RTD) != 0)
{
    printf("Failed to set up channel temperature type.\n");
    return -1;
}
if (ulAISetConfig(daqDeviceHandle[devN], AI_CFG_CHAN_SENSOR_CONNECTION_TYPE, i, rtdType[devN]) != 0)
{
    printf("Failed to set up channel temperature type.\n");
    return -1;
}
```
Neither of those call trigger any errors, but when reading the channel I get the error 85: ERR_OPEN_CONNECTION and get the result -9999.
If I use those config but wire a thermocouple it works.

Am I doing anything wrong?

sbazaz, 2023-05-26:
The USB-TEMP's channel type can only be configured through InstaCal. However, I'm not sure why you don't get an error when you call ulAISetConfig.

thojDufour, 2023-05-31:
Weird, even weirder that such a big, and known issue has not been corrected by now. I mean the fact that the USB-Temp can only be configured with InstaCal.