mccdaq/uldaq #56: Timer's status getter returns incorrect state

oswinrodrigues, 2023-06-16:
I'm using the `uldaq` Python library with a USB-CTR04 device. 

Whenever I call the `get_pulse_out_status` [method](https://www.mccdaq.com/PDFs/Manuals/UL-Linux/python/api.html#uldaq.TmrDevice.get_pulse_out_status), it always gives me `TmrStatus.RUNNING`, regardless of the actual state of the timer. 

```python
Python 3.10.6 (main, May 29 2023, 11:10:38) [GCC 11.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import uldaq
>>> devices = uldaq.get_daq_device_inventory(uldaq.InterfaceType.USB)
>>> daq = uldaq.DaqDevice(devices[0])
>>> daq.connect()
>>> tmr.get_pulse_out_status(0)
<TmrStatus.RUNNING: 1>
>>> tmr.pulse_out_stop(0)
>>> tmr.get_pulse_out_status(0)
<TmrStatus.RUNNING: 1>
>>> tmr.pulse_out_start(0, 100, 0.5, 0, 0.0, uldaq.TmrIdleState.LOW, uldaq.PulseOutOption.DEFAULT)
(100.0, 0.5, 0.0)
>>> tmr.get_pulse_out_status(0)
<TmrStatus.RUNNING: 1>
>>> tmr.pulse_out_stop(0)
>>> tmr.get_pulse_out_status(0)
<TmrStatus.RUNNING: 1>
```

I believe this is a bug. The method should return `TmrStatus.IDLE` when the pulses have ceased.

sbazaz, 2023-08-11:
Please submit the issues through the MCC technical [support page](https://www.mccdaq.com/support/support_form.aspx)

Thank you

oswinrodrigues, 2023-08-31:
This was already done here: https://forum.digilent.com/topic/24751-precise-control-of-timers-with-python-library/

I just didn't know which was the more fitting place to post, so did it in both. Posting this here in case anyone else is looking for the same info.

(Note: the issue wasn't fully resolved. MCC support was not being responsive. But it sounds like the USB-CTR04 board just can't do what I'm looking for. The documentation was just plain misleading.)

oswinrodrigues, 2023-08-31:
So what are issues on this Github project meant to be used for? @sbazaz