mccdaq/uldaq #57: Refresh rate issue

thojDufour, 2023-10-05:
While using [ulDaqInScan](https://files.digilent.com/manuals/UL-Linux/c/group___sync_io.html#gad04454d9ae318d7f9959dae3d5bdab4f) every 10ms (100 times per second) the [currentScanCount](https://files.digilent.com/manuals/UL-Linux/c/struct_transfer_status.html#a3495754bbfd64a1826527c7f559fc1fc) increase by a number that does not necessarily reflect the requested sample rate.

For example at 10 [kS/s] the expected amount of scan is 100, but the function returns a [currentScanCount](https://files.digilent.com/manuals/UL-Linux/c/struct_transfer_status.html#a3495754bbfd64a1826527c7f559fc1fc) that is 128 count higher than 0.01 [s] before.
For 12.8 [kS/s] the function still returns a [currentScanCount](https://files.digilent.com/manuals/UL-Linux/c/struct_transfer_status.html#a3495754bbfd64a1826527c7f559fc1fc) that is 128 count higher than 0.01 [s] before. Which in this case makes sense.

Same issue between 25 [kS/s] and 25.6[kS/s].

At high sample rate the scan count appear to work fine for multiple of the polling rate and 128. This does not appear to be the case at lower sample rate. I was not able to find a rule that would give me a sample rate that would return the correct amount of scan counts.

FaSoa, 2023-10-05:
Hello @thojDufour .
Which MCC daq device are you using?

sbazaz, 2023-10-11:
Are you using the continuous option (SO_CONTINUOUS)? If that the case, then that's the expected behavior. the scan samples are transferred in multiples of block size from the scan buffer. If you don't like this behavior, you can use the single I/O option, however, you might get buffer overrun error at high scan rates.

thojDufour, 2023-10-11:
Hello @FaSoa ,
Ich benutze beide der USB-1808 und USB-1808X.

@sbazaz  Yes, I'm indeed using the SO_CONTINUOUS option.
Is there anywhere in the documentation where I can find out the block size of the scan buffer for specific devices?

Let's say for example I set the sampling rate to 10kS/s and I'm polling 100 times a second. I receive therefore 128 samples. But there is only 100 new samples that have happened since last poll. Is it correct to assume that the first 28 samples of the last poll are the same as the last 28 polls from the previous poll?

sbazaz, 2023-11-30:
Please submit the issues through the MCC technical [support page](https://forum.digilent.com/forum/39-measurement-computing-mcc/)

Thank you

closed 2023-11-30 by sbazaz