mccdaq/uldaq #58: Refresh rate issue 2

thojDufour, 2023-10-24:
Using the example code for DaqInScan.c with as little modification as possible to demonstrate the issue, I noticed that for a sample rate of 12800 samples per seconds, and calling the function ulDaqInScanStatus 100 times per second, I get most of the time 128 samples. A few times per second I receive 256 samples, and more rarely 0 samples.

I tried that after noticing this issue in a more complex software were the computer time is logged alongside the data, and the software calls the ulDaqInScanStatus function every 10000 micro seconds +- 2 microseconds. The error repeat too often to be due to the inaccuracy of the call.

This cause some serious issues on a more complex software where I synchronize multiple devices and where I want to log the data.

Below the modified example code as .txt to be able to upload it.
[DaqInScan.txt](https://github.com/mccdaq/uldaq/files/13117091/DaqInScan.txt)

thojDufour, 2023-10-24:
I forgot to say, that I'm using USB-1808 and USB-1808X.

sbazaz, 2023-11-30:
Please submit the issues through the MCC technical [support page](https://forum.digilent.com/forum/39-measurement-computing-mcc/)

Thank you

closed 2023-11-30 by sbazaz